macdylan/sm2uploader #3: Instructions are terrible.

Learoy1, 2023-04-04:
Most of us aren't coders, we can't guess what to do with the files please provide adequate instructions.

apphead, 2023-04-04:
Being friendly and a thank would have been more appropriate. Guess you couldn't code it yourself,  right 👍?

But since you asked, there are good instructions here:
https://github.com/leandrolima-nyc/SnapmakerJ1#j1-auto-upload-by-kanocz

macdylan, 2023-04-05:
I'm sorry that this is a hobby project, and I didn't spend much time at the instructions, because it's a command-line tool that doesn't need to be explained to most users who think they need it.

Btw, [J1-Upload](https://github.com/kanocz/j1upload) is a great project that I use it almost every day ❤️

closed 2023-04-05 by macdylan