macdylan/sm2uploader #4: Octoprint settings?

apphead, 2023-04-23:
What are the right settings to use the simulated octoprint?
Without IP, the upload button is not displayed
With correct IP, the upload works, but PS throws an error not being able to upload (connection test fails too, that is to be expected)
What settings would you recommend?
[image]
[image]

macdylan, 2023-04-24:
If IP:Port is correct, opening it in the browser will display a 404 instead of any other error. Please make sure the port number is correct.

apphead, 2023-04-24:
Ah, OK. port 8888 seems to be correct default.
Now, "RESOLVING ADDRESS" pops up and stays there forever. I prefer the "error" edition  ;-)
[image]

macdylan, 2023-04-24:
Please try this:
[image]
[image]

macdylan, 2023-04-24:
If you are running sm2up on another computer/nas/octopi, please make sure that the firewall is not blocking that port.

prusaslicer/s3d/ideamaker via OctoPrint protocol -> sm2up Dummy-OctoPrint-Service -> Snapmaker Printer via SM Protocols

That is how it works.

closed 2023-06-28 by macdylan