macdylan/sm2uploader #8: Discovery doesn't work if subnet is not /24

KyGunsAndRadio, 2023-09-03:
I normally run printers on a separate subnet configured as 10.10.10.0/25 (255.255.255.128). My client PC is also attached directly to that subnet. But when running sm2uploader, discovery mode would not find my printer. I temporarily changed that subnet to /24 (255.255.255.0) and discovery started working. I'm guessing it's because it's trying to broadcast from 10.10.10.255, which isn't valid on a /25 subnet. Perhaps this is a limitation of UDP discovery?

I was able to use the -host switch and force the IP address, but in that case it didn't identify the printer model. Not sure if that's an issue or not.

macdylan, 2023-09-04:
Yes, currently it will only broadcast to address 255 under the /24 subnet, which I think is the same as luban. I will test and fix it.

macdylan, 2023-09-05:
I'v published v2.3, Please let me know if it is resolved.

KyGunsAndRadio, 2023-09-08:
Wow, thanks for the quick response! I tried 2.3. This time it seems to be broadcasting on 10.0.0.0 and failed to find anything. I also just tried changing my subnet back to /24, but it did the same thing. Is it possible to determine what the current configured subnet is and set the broadcast accordingly? May be more trouble than it's worth, I doubt too many people are running networks like mine.

[image]

macdylan, 2023-09-10:
Can you post the result of `ip addr` ?

KyGunsAndRadio, 2024-03-25:
I know longer have this particular network config in place, so I can't help support the change. As a result, I closed this issue. If someone in the future should need this, they can re-open or create a new issue.

closed 2024-03-25 by KyGunsAndRadio