macdylan/sm2uploader #9: Pressing YES is still required

Fuchschr11, 2023-10-05:
Hi,
great tool ;-)

Using version 2.2 everything works good, but I have to press YES everytime. I have an A350.

`Air-M1-2021:~ fuchschr$ /Users/fuchschr/Library/CloudStorage/Dropbox/3D_Druck/Snapmaker_tools/sm2uploader/sm2uploader -octoprint 127.0.0.1:8844 -host 10.0.0.113
2023/10/05 11:27:12 Discovering ...
2023/10/05 11:27:16 Printer IP: 10.0.0.113
2023/10/05 11:27:16 Starting OctoPrint server on 127.0.0.1:8844 ...
2023/10/05 11:27:16 Server started, now you can upload files to http://127.0.0.1:8844
2023/10/05 11:29:50 Request GET / completed in 106.916µs
2023/10/05 11:29:50 Request GET /favicon.ico completed in 29.916µs
2023/10/05 11:33:52 Request GET /api/version completed in 27.167µs
2023/10/05 11:34:21 Request GET /api/version completed in 9.375µs
2023/10/05 11:34:22 >>> Please tap Yes on Snapmaker touchscreen to continue <<<`

Any ideas?

greetings
Christian

macdylan, 2023-10-09:
To check if a directory is writable, try putting sm2up in the /tmp directory, which will generate a hosts.yml file with the last connection token.

Fuchschr11, 2023-10-11:
Hi, thanks but the directory is writeable. You see in the video it will create a hosts.yaml with no printer information.
see video https://lernenmitsystem-my.sharepoint.com/:v:/g/personal/christian_fuchs_lernenmitsystem_onmicrosoft_com/ER11NN7QPPlFuHBVHiOg2x8B8c3UtaFY0TWMYhwlG5M3Ww?e=jL4zLs 

Thanks for you help

greetings
Christian

macdylan, 2023-10-12:
Can you try to remove the -host parameter.

or write to the hosts.yaml file:
```
printers:
    - ip: 10.0.0.113
      id: AA350
      model: Snapmaker 2 Model A350
      token: 1111-1111
      sacp: false
```

`... -host AA350`, That token will be updated after the first connection

Fuchschr11, 2023-10-12:
Hi,

hosts.yaml looks now like this:

`printers:
    - ip: 10.0.0.113
      id: AA350
      model: Snapmaker 2 Model A350
      token: cf2d1b04-4c7e-4b30-84eb-cd68f7972251
      sacp: false
`
The token was written in the host.yaml.

It will still ask for "YES" on the snapmaker.

`'/Users/fuchschr/Library/CloudStorage/Dropbox/3D_Druck/Snapmaker_tools/sm2uploader/sm2uploader' -octoprint 127.0.0.1:8844
2023/10/12 18:56:23 Discovering ...
2023/10/12 18:56:27 Printer IP: 10.0.0.113
2023/10/12 18:56:27 Printer Model: Snapmaker 2 Model A350
2023/10/12 18:56:27 Starting OctoPrint server on 127.0.0.1:8844 ...
2023/10/12 18:56:27 Server started, now you can upload files to http://127.0.0.1:8844
2023/10/12 18:56:42 Request GET /api/version completed in 24.709µs
2023/10/12 18:56:42 >>> Please tap Yes on Snapmaker touchscreen to continue <<<
`

Any ideas?

greetings
Christian

macdylan, 2023-10-13:
Please add `-debug` flag to see what the output is. Also, you can run this once in the terminal: `sm2uploader /path/to/your/file.gcode`

Fuchschr11, 2023-10-13:
Hi, 

thanks, see log in attachment.

greetings
Christian
[sm2uploader_log.txt](https://github.com/macdylan/sm2uploader/files/12896811/sm2uploader_log.txt)

Fuchschr11, 2023-10-13:
Hi, after playing around with -debug, it seems to work now.

Last login: Fri Oct 13 19:41:20 on ttys003

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
MacBook-Air-M1-2021:~ fuchschr$ '/Users/fuchschr/Library/CloudStorage/Dropbox/3D_Druck/Snapmaker_tools/sm2uploader/sm2uploader' -octoprint 127.0.0.1:8844
2023/10/13 19:43:45 Discovering ...
2023/10/13 19:43:49 Printer IP: 10.0.0.113
2023/10/13 19:43:49 Printer Model: Snapmaker 2 Model A350
2023/10/13 19:43:49 Starting OctoPrint server on 127.0.0.1:8844 ...
2023/10/13 19:43:49 Server started, now you can upload files to http://127.0.0.1:8844
2023/10/13 19:44:00 Request GET /api/version completed in 35.583µs
2023/10/13 19:44:00 G-Code fixed
2023/10/13 19:44:02   - HTTP sending 100.3%
2023/10/13 19:44:03 Upload finished: red.gcode [907.2 KB]
2023/10/13 19:44:03 Request POST /api/files/local completed in 2.88458775s

I had to press "YES,..... " once.

I will observe it.

macdylan, 2023-11-02:
I will close this PR, please feel free to reply if the issue happens again.

closed 2023-11-02 by macdylan