macdylan/sm2uploader #11: Broadcast address calculation was broken in 2c99a44

lestrrat, 2023-11-01:
The code [here](https://github.com/macdylan/sm2uploader/commit/2c99a446f4d8e77c9b5f0b1c05397f8036738af1#diff-068065d268947533fdd6a35cb58db82ead3ee806bd27029c62d9f43ee4753457R99) breaks broadcast address calculation, as applying the mask simply yields in the right most octet as 0.

Here's a fix. Let me know if you want me to create PR for it

```diff
diff --git a/discover.go b/discover.go
index 3ecd630..78ad5e6 100644
--- a/discover.go
+++ b/discover.go
@@ -1,6 +1,7 @@
 package main
 
 import (
+       "encoding/binary"
        "fmt"
        "log"
        "net"
@@ -95,8 +96,15 @@ func getBroadcastAddresses() ([]string, error) {
                        continue
                }
                for _, addr := range addrs {
-                       if n, ok := addr.(*net.IPNet); ok && !n.IP.IsLoopback() && n.IP.To4() != nil {
-                               baddrs = append(baddrs, n.IP.Mask(n.IP.DefaultMask()).String())
+                       if n, ok := addr.(*net.IPNet); ok && !n.IP.IsLoopback() {
+                               if v4addr := n.IP.To4(); v4addr != nil {
+                                       // convert all parts of the masked bits to its maximum value
+                                       // by converting the address into a 32 bit integer and then
+                                       // ORing it with the inverted mask
+                                       baddr := make(net.IP, len(v4addr))
+                                       binary.BigEndian.PutUint32(baddr, binary.BigEndian.Uint32(v4addr)|^binary.BigEndian.Uint32(n.IP.DefaultMask()))
+                                       baddrs = append(baddrs, baddr.String())
+                               }
                        }
                }
        }
```

PS: I'd also like to create a PR that removes duplicate broadcast addrs, if it's OK with you.

macdylan, 2023-11-02:
LGTM, please create a PR, thank you.

lestrrat, 2023-11-02:
@macdylan done: #12

PR #12 (2023-11-02): Fix broadcast address calculation
Calculate the broadcast address by converting the v4 address to a uint32 first, and the OR'ing it with the inverted netmask before converting it back into octets stored in net.IP

fixes #11
closed 2023-11-02 by macdylan