macdylan/sm2uploader #13: Auto discovery not working

adammhaile, 2023-12-31:
I have a single J1S on my network and am trying to use sm2uploader from OrcaSlicer to upload my files. Unfortunately I'm having no luck even when supplying the IP address directly.  

When I don't provide an IP it never finds any printer, even though I can ping it and Luban finds it perfectly fine.
When I do provide an IP it still tries to auto discover (I feel like it shouldn't be doing this) and it seems like it connects but then throws an SACP error. See log below.

I'm running firmware v2.6.2 on the printer which it tells me is the latest.

Any thoughts greatly appreciated. 

```
PS C:\Users\admin\Downloads\sm2uploader-win64.exe> .\sm2uploader-win64.exe -debug
2023/12/31 12:45:06 -- Debug mode
2023/12/31 12:45:06 Discovering ...
2023/12/31 12:45:06 -- Discovering on 169.254.255.255:20054
2023/12/31 12:45:10 -- Discovering on 10.255.255.255:20054
2023/12/31 12:45:14 -- Discovering on 169.254.255.255:20054
2023/12/31 12:45:18 -- Discovering on 169.254.255.255:20054
2023/12/31 12:45:22 -- Discovering on 169.254.255.255:20054
2023/12/31 12:45:26 -- Discovering on 169.254.255.255:20054
2023/12/31 12:45:30 -- Discovering on 169.254.255.255:20054
2023/12/31 12:45:34 -- Discovered 0 printers
2023/12/31 12:45:34 No printers found
2023/12/31 12:45:34 -- Saved known hosts: C:\Users\admin\Downloads\sm2uploader-win64.exe\hosts.yaml
PS C:\Users\admin\Downloads\sm2uploader-win64.exe> .\sm2uploader-win64.exe -debug -host 10.0.1.57 "C:\Users\admin\Downloads\test_upload.gcode"
2023/12/31 12:51:50 -- Debug mode
2023/12/31 12:51:50 Discovering ...
2023/12/31 12:51:50 -- Discovering on 169.254.255.255:20054
2023/12/31 12:51:54 -- Discovering on 10.255.255.255:20054
2023/12/31 12:51:58 -- Discovering on 169.254.255.255:20054
2023/12/31 12:52:02 -- Discovering on 169.254.255.255:20054
2023/12/31 12:52:06 -- Discovering on 169.254.255.255:20054
2023/12/31 12:52:10 -- Discovering on 10.255.255.255:20054
2023/12/31 12:52:14 -- Discovering on 169.254.255.255:20054
2023/12/31 12:52:18 -- Discovered 0 printers
2023/12/31 12:52:18 Printer IP: 10.0.1.57
2023/12/31 12:52:18 Uploading file 'test_upload.gcode' [245.7 KB]...
2023/12/31 12:52:22 -- SACP_connect got:
&{0 1 1 32056 172 165 [0 0 0 0 0]}
2023/12/31 12:52:22 got command set/id 172/165 but expected 1/5
2023/12/31 12:52:22 -- Updated printer: @10.0.1.57 -
2023/12/31 12:52:22 -- Saved known hosts: C:\Users\admin\Downloads\sm2uploader-win64.exe\hosts.yaml
PS C:\Users\admin\Downloads\sm2uploader-win64.exe> ping 10.0.1.57

Pinging 10.0.1.57 with 32 bytes of data:
Reply from 10.0.1.57: bytes=32 time=42ms TTL=64
Reply from 10.0.1.57: bytes=32 time=57ms TTL=64
Reply from 10.0.1.57: bytes=32 time=75ms TTL=64
Reply from 10.0.1.57: bytes=32 time=88ms TTL=64

Ping statistics for 10.0.1.57:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 42ms, Maximum = 88ms, Average = 65ms
```

macdylan, 2023-12-31:
> 2023/12/31 12:52:22 got command set/id 172/165 but expected 1/5

Please upgrade the firmware and restart the printer.

adammhaile, 2023-12-31:
Weird... yup, that works. What's stranger though is I had not run Luban to test until *after* it failed to work the first time (2 minutes after first startup of printer) with sm2uploader.
One question... it still tries to discover printers on the network even with `-host` and still fails. I thought that providing `-host` disabled discovery? j1uploader does work with auto discovery... but nothing I can do with sm2uploader gets auto to work.

TheBum, 2024-05-28:
I'm having a similar issue on an M2 MacBook Pro when communicating with my Artisan. The app connects if I specify the host explicitly, but not if I omit that option. I'm running version 2.8 of sm2uploader. Sometimes I get this:
`2024/05/28 11:34:43 Discovering ...
2024/05/28 11:34:43 Error discovering on 192.168.68.255: write udp4 0.0.0.0:59967->192.168.68.255:20054: sendto: no route to host
2024/05/28 11:34:43 No printers found`
whereas other times, I simply get this:
`2024/05/28 11:36:46 Discovering ...
2024/05/28 11:36:50 No printers found`