macdylan/sm2uploader #14: Upload and start print automatically

nailgilaziev, 2024-02-15:
When using simulated OctoPrint server to send g-code from slicer directly like this

`./sm2uploader-darwin-arm64 -octoprint 127.0.0.1:8844  -host 192.168.1.174`

There are two options:
- Upload
- Upload and Print

[image]

Is it possible that printing will start automatically after sending the code by pressing second option?

macdylan, 2024-02-19:
I'm still trying to implement it, but the protocols for Snapmaker are not the same in each model, so it's complicated.

Exc021, 2024-02-22:
+1 this would be very nice

gcreative, 2024-03-23:
@macdylan can you explain to me how to research the protocol? how to organize the connection? is there any documentation? I know how to use wireshark a little and I know how to program; maybe I can help

macdylan, 2024-03-26:
The Snapmaker protocol v0 is HTTP-based, while v1 is completely private and is documented [here](https://github.com/Snapmaker/Snapmaker-SACP/tree/main/docs), but not completely, you need the luban [source code](https://github.com/Snapmaker/Luban/tree/main/src/server/services/machine) to analyze.

perfran, 2024-07-17:
Hi, 
I second this improvement, sending the print command to the printer (Artisan in my case) would be very useful.
If it can help, I know it works with Cura (snapmaker plugin). Maybe looking in that code can give useful insights.
https://github.com/Snapmaker/SnapmakerCuraPlugin
Thanks for the beautiful work, @macdylan !

macdylan, 2024-07-19:
Sorry guys, I am currently working on another project in my spare time, so this feature will not be implemented for now. If anyone is interested in improving it, please submit a PR.

chrisns, 2025-01-09:
https://github.com/vampi62/jeedom_snapmaker/blob/dc0f7d71ca67607f9da592e0689321f115873320/resources/snapmakerd/snapmakerd.py#L85-L114 covers it reasonably specifically

in short looks like send the file to `prepare_print` instead of `upload` then hit `start_print`  with the filename

I'll try remembering how to write go and making a PR in between print jobs :)