macdylan/sm2uploader #15: When printing with only the right nozzle, the nozzle temperature from the second layer is set to 0

monohoshi, 2024-02-25:
Printed under general conditions with right nozzle only. 
When I set up as shown in the figure, the first layer prints correctly, but from the second layer the nozzle temperature is set to 0 and printing fails.
Also, changes in bed temperature were not recognized.
Temperatures on other layers do not seem to be reflected correctly.

[image]

macdylan, 2024-03-01:
I can't reproduce it on J1, can you provide more information? it would be more helpful if you could provide gcode or 3mf file, thanks for your FDBK.

macdylan, 2024-03-07:
It seems that you are using the old profile, please try the latest profiles released in 2.7.1, there are some temperature controls for dual extruder.

[image]
[image]

closed 2024-04-28 by macdylan