macdylan/sm2uploader #19: Wrong temperatures are set for IDEX J1

obertini78, 2024-04-07:
Hi,

The waiting and printing temperatures are wrongly set on dual material projects in OrcaSlicer with Prime Tower enabled.

**How to reproduce**

1. In OrcaSlicer (v1.9.1 to 2.0.0), setup a new project with two different filaments with different printing temperatures (255° for T0 and 220°C for T1 with standby temperatures set to 178°C and 154°C respectively).
2. Add two basic shapes to the print plate.
3. Set the first plate to Filament 1 and set the second shape to Filament 2.
4. In Others, Enable Prime Tower. If Prime Tower is not enabled, this bug is not visible.
5. Slice and upload to the printer via wifi using the octoprint server.

Here is what the sliced project looks like:
[image]

**Actual results:**
The printer starts to print with T0 (255°C) and T1 is waiting (154°C).
Then the first tool change happens.
S1 - T0=>T1:
              T0=>178°C // waiting...
              T1=>220°C  // printing... All good so far
S2 - T1=>T0:
              T0=>255°C  // printing...
              T1=>255°C // waiting... Incorrect temperature. Should be 154°C
S3 - T0=>T1:
              T0=>220°C // waiting... Incorrect temperature. Should be 178°C
              T1=>255°C // printing... Incorrect temperature. Should be 220°C
S4 - T1=>T0:
              T0=>220°C // printing... Incorrect temperature. Should be 255°C
              T1=>154°C // waiting...
S5 - T0=>T1:
              T0=>220°C // waiting... Incorrect temperature. Should be 178°C
              T1=>220°C // printing...
S6 - T1=>T0:
              T0=>220°C // printing... Incorrect temperature. Should be 255°C
              T1=>0°C // waiting... correct as this extruder will no longer be used

**Expected results:**
All print and waiting temperatures are set correctly.

Looking into the sm2uploader processed g-code, one can see the following (line 862 of Cube_wifi.gcode).
```
; CP TOOLCHANGE UNLOAD
; LINE_WIDTH: 0.6
; LINE_WIDTH: 0.5
M104 S255
; Cooling park
G1 E-2
;***** Update: 20230730

; Change T1 -> T0 (layer 1
; layer
M106 S0
T0

;(Fixed: remove cooldown: M104 T1 S154 ;standby T1)

; set bed temp: PLA(60) -> PETG(70)
M140 S60

M2000 S200 V350 A11000 ;quick switch extruders, S:200 mode/V:speed/A:acceleration
M109 T0 S255 C3 W1 ;wait T0
```

There is a _M104 S255_ command that is added by OrcaSlicer only when the Prime Tower is enabled.
The _;(Fixed: remove cooldown: M104 T1 S154 ;standby T1)_ removes the waiting temperature instruction added from my "Change Filament G-code" section in OrcaSlicer. 

For the IDEX J1, the g-code should rather looks like
```
; CP TOOLCHANGE UNLOAD
; LINE_WIDTH: 0.6
; LINE_WIDTH: 0.5
**;(Fixed: remove purge temperature M104 S255)**
; Cooling park
G1 E-2
;***** Update: 20230730

; Change T1 -> T0 (layer 1
; layer
M106 S0

T0

**M104 T1 S154 ;standby T1**

; set bed temp: PLA(60) -> PETG(70)
M140 S60

M2000 S200 V350 A11000 ;quick switch extruders, S:200 mode/V:speed/A:acceleration
M109 T0 S255 C3 W1 ;wait T0
```

You can find attached the OrcaSlicer saved 3mf project file as well as the g-code files. 'Cube_usb.gcode' native from OrcaSlicer and 'Cube_wifi.gcode' processed by sm2uploader.

[sm2uploader_wrong_temperatures.zip](https://github.com/macdylan/sm2uploader/files/14898652/sm2uploader_wrong_temperatures.zip)

macdylan, 2024-04-08:
Please try v2.7(pre-release) and let me know if it works, Thanks.

obertini78, 2024-04-08:
Worked like a charm on my test print :-)
Thanks. Will continue to use / test.

DimitryNechaev, 2024-04-28:
I experienced the same issue trying to figure out why mu TPU clogs when printing along with PLA. It was switching TPU to 220 on some layer early in the print. It seems to be fixed now.

closed 2024-04-28 by macdylan