macdylan/sm2uploader #22: Error: G-Code fix error(ignored): Invalid G-Code file

hsai1020, 2024-06-30:
I'm using the latest firmware (2.7.2) on the J1, and am trying to send G-code over the Wifi connection. I seem to be getting the above error for multiple models and from multiple platforms (RPi5, Windows 11) when I send the code from PrusaSlicer. I'm using the printer profile from [leandrolima-nyc](https://github.com/leandrolima-nyc/SnapmakerJ1). Below is the message in the OctoPrint emulator.

---
2024/06/29 22:00:03 -- Debug mode: v2.8
2024/06/29 22:00:03 Discovering ...
2024/06/29 22:00:03 -- Discovering on 192.168.1.255:20054
2024/06/29 22:00:03 -- Discover got 46 bytes J1A400@192.168.1.169|model:Snapmaker J1|SACP:1
2024/06/29 22:00:07 -- Discovered 1 printers
2024/06/29 22:00:07 Printer IP: 192.168.1.169
2024/06/29 22:00:07 Printer Model: Snapmaker J1
2024/06/29 22:00:07 Starting OctoPrint server on localhost:8844 ...
2024/06/29 22:00:07 Server started, now you can upload files to http://127.0.0.1:8844
2024/06/29 22:00:17 Request GET /api/version completed in 9.241µs
2024/06/29 22:00:18 -- SACP_connect got:
&{0 1 1 42253 21 161 [0 40 255]}
2024/06/29 22:00:18 -- SACP_connect got:
&{0 2 1 1 1 5 [0 0 0]}
2024/06/29 22:00:18 -- Connected to printer
2024/06/29 22:00:19 G-Code fix error(ignored): Invalid G-Code file.
2024/06/29 22:00:21 -- Got reply from printer: &{0 2 0 1 176 2 [0]}
2024/06/29 22:00:21 -- Upload finished
2024/06/29 22:00:21 Upload finished: centrifuge_rack_2mm_0.3_T1PLA+T2PLA.gcode [0 B]
2024/06/29 22:00:21 Request POST /api/files/local completed in 4.249929859s
---

I can confirm that downgrading to v2.6 solved the issue. v2.7 shows the same issue, so somewhere between v2.6 and v2.7 the issue was introduced.

macdylan, 2024-06-30:
Can you try PrusaSlicer [v2.8.0](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0) ? I added snapmaker profiles for it.

hsai1020, 2024-07-05:
Sorry for the belated response. I'm using the ARM version (https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases) on RPi5 and it seems to be showing the same error.

```
Jul 05 06:41:52 raspberrypi sm2uploader-linux-arm7[18148]: 2024/07/05 06:41:52 Request GET /api/version completed in 26µs
Jul 05 06:41:54 raspberrypi sm2uploader-linux-arm7[18148]: 2024/07/05 06:41:54 G-Code fix error(ignored): Invalid G-Code file.
Jul 05 06:41:55 raspberrypi sm2uploader-linux-arm7[18148]: 2024/07/05 06:41:55   - SACP sending 100.0%
Jul 05 06:41:56 raspberrypi sm2uploader-linux-arm7[18148]: 2024/07/05 06:41:56 Upload finished: J1-ptfe-adapter v5 (1)_0.2_T1PLA+T2PETG.gcode [0 B]
Jul 05 06:41:56 raspberrypi sm2uploader-linux-arm7[18148]: 2024/07/05 06:41:56 Request POST /api/files/local completed in 3.199039794s

```

hsai1020, 2024-09-25:
Sorry, it looks like I had a wrong version loaded for sm2uploader... PrusaSlicer v2.8.0 + sm2uploader v2.8 can upload gcodes successfully to J1. Thank you so much and sorry for the trouble!

closed 2024-09-25 by hsai1020