macdylan/sm2uploader #23: 强化擦料塔功能似乎有一些问题 Some questions about reinforce the prime tower

funnygeeker, 2024-08-23:
机器：Snapmaker J1s
切片：Orca
我在打印 PLA + PETG 时，观察到经过 SM2Uploader 处理的擦料塔打印时似乎在个别空驶或打印的情况下，由于挤出过多会导致喷嘴与模型发生碰撞，碰撞时可明显听到噪音和感到震动，在擦料塔高到一定程度时，如果不对擦料塔外壁进行特别限速（50mm/s）可能会使擦料塔更容易倒塌。似乎应该稍微降低挤出量或者添加Z抬升以达到最佳效果

另外：在耗材丝的可溶性材料中勾选，可以使擦料塔在异材料打印的时候打印更稳定，但是上述的问题可能会一定程度影响到同材料多色打印时的稳定性。

English (The translation was used.)
Machine: Snapmaker J1s
Slicer: Orca
While printing with PLA + PETG, I observed that the wiping tower processed by SM2Uploader seems to occasionally experience collisions between the nozzle and the model due to excessive extrusion during idle or printing situations. These collisions can be heard with noticeable noise and felt as vibrations. When the wiping tower reaches a certain height, failure to impose special speed restrictions (50mm/s) on the outer wall of the tower may lead to it collapsing more easily. It appears that slightly reducing the extrusion amount or adding Z-lift may achieve optimal results.

Additionally, checking soluble materials in the filament can enhance the stability of the wiping tower during multi-material printing with different materials. However, the above issue may somewhat affect the stability during multi-color printing with the same material.

macdylan, 2024-08-24:
> 擦料塔打印时似乎在个别空驶或打印的情况下，由于挤出过多会导致喷嘴与模型发生碰撞

你可以试试先关闭 force tower 功能: `-noreinforcetower;`
[image]
[image]

在下周发布新的 Snapmaker Profiles for OrcaSlicer 之后，会更新 sm2uploader & smfix，今后的版本都将默认关闭 preheat 和 reinforecetower 功能，来配合新版参数。

> 如果不对擦料塔外壁进行特别限速（50mm/s）可能会使擦料塔更容易倒塌

OrcaSlicer 的版本是多少？旧版本在擦料塔的速度控制上有错误，已经发布了修正 [#6044](https://github.com/SoftFever/OrcaSlicer/pull/6044) 并且已经包含在最新的版本中

funnygeeker, 2024-08-24:
OrcaSlicer Version: 2.1.1
Thank you for your reply :P

macdylan, 2024-08-24:
v2.1.0以后使用单独的擦料塔速度设置:

[image]

funnygeeker, 2024-08-24:
是的，我按照这个设置了，但是里边似乎还有一个提示说到了：无论这个参数设置如何，擦料塔外壁都会使用内壁速度。
Yes, I set it according to this, but there seems to be another prompt inside: No matter how this parameter is set, the outer wall of the wiping tower will use the inner wall speed.

funnygeeker, 2024-08-24:
[image]

funnygeeker, 2024-08-24:
目前这个问题似乎还没有修正,我正在等待新版本.
Currently this problem doesn't seem to be fixed, I'm waiting for the new version.
（Orca: V2.1.1）

funnygeeker, 2024-08-24:
> > 擦料塔打印时似乎在个别空驶或打印的情况下，由于挤出过多会导致喷嘴与模型发生碰撞
> 
> 你可以试试先关闭 reinforce tower 功能：`-noreinforcetower;` <img alt="image" width="614" src="https://private-user-images.githubusercontent.com/331506/361126071-f6abe63f-b95e-4539-9fe3-4e275af45a5a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ0ODczNzYsIm5iZiI6MTcyNDQ4NzA3NiwicGF0aCI6Ii8zMzE1MDYvMzYxMTI2MDcxLWY2YWJlNjNmLWI5NWUtNDUzOS05ZmUzLTRlMjc1YWY0NWE1YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyNFQwODExMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YjA2MDZlMmJjM2MyYzBkOTU3NGZiZGM5MTkxYmZhMDY0ZDMwMWNlYjdlNjY2Yjg0Nzk2MmZjZDFhY2IyZTAxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.X9N9T392Wt_1vflsy1Xwn0TFsNObPPWb37lNnQZIuLU"> <img alt="image" width="532" src="https://private-user-images.githubusercontent.com/331506/361126093-6de29f60-1509-4395-be36-7cbca5238c50.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ0ODczNzYsIm5iZiI6MTcyNDQ4NzA3NiwicGF0aCI6Ii8zMzE1MDYvMzYxMTI2MDkzLTZkZTI5ZjYwLTE1MDktNDM5NS1iZTM2LTdjYmNhNTIzOGM1MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyNFQwODExMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03Y2QyMmM1Yzg1N2FjODkxZDljNzg1MDBiODcxMThiZDZlNjA0ZjRkZjY2ZjQ1NzNhMDg2MjkyOGVlNDFkNTE3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QyEZMHbbn9apYBQ0hgmeynyjrWWkJpEYDT_rZt8kQy8">
> 
> 在下周发布新的Snapmaker Profiles for OrcaSlicer 之后，会更新 sm2uploader & smfix，今后的版本都将默认关闭 preheat 和 reinforecetower 功能，来配合新版参数。
> 
> > 如果不对擦料塔外壁进行特别限速（50mm/s）可能会使擦料塔更容易倒塌
> 
> OrcaSlicer 的版本是多少？旧版本在擦料塔的速度控制上有错误，已经发布了修正 [#6044](https://github.com/SoftFever/OrcaSlicer/pull/6044) 并且已经包含在最新的版本中

您提交的PR晚于这个版本的发布月份，可能需要等到下一个版本的 Release 发布才能正常使用.
The PR you submitted is later than the Release month of this version, and you may need to wait until the next version of Release is released before it can be used normally.

funnygeeker, 2024-08-24:
感谢您对社区的贡献
Thank you for your contribution to the community

closed 2024-08-24 by funnygeeker